woodemi/quick_usb #12: Would you consider adding Null Safety and iOS/iPadOS support?

AriaShishegaran, 2021-03-10:
@Sunbreak hi there, since you have worked on the desktop release as well as MacOS I wanted to know whether you will continue your work on completing the package to support both null safety and add support for iOS?

Sunbreak, 2021-03-10:
Null-Safety is coming. But I don't think I'll take into iOS/iPadOS because of https://mfi.apple.com/

digitalkirkley, 2021-03-16:
Would you be able to update the Ffi to 1.0.0 please? I was going to test this on latest Flutter Linux desktop but has an error for the getter 'addressOf' isn't defined for the class Utf8.  Thank you.

Sunbreak, 2021-03-16:
> Would you be able to update the Ffi to 1.0.0 please? I was going to test this on latest Flutter Linux desktop but has an error for the getter 'addressOf' isn't defined for the class Utf8. Thank you.

Could you post another issue for null-safety?

Sunbreak, 2021-03-17:
> Would you be able to update the Ffi to 1.0.0 please? I was going to test this on latest Flutter Linux desktop but has an error for the getter 'addressOf' isn't defined for the class Utf8. Thank you.

Could you try https://github.com/woodemi/quick_usb/pull/13?

digitalkirkley, 2021-03-17:
Yes sure - will give this a go; brilliant thank you.

Sunbreak, 2021-04-17:
Could you try #17 ?

Sunbreak, 2021-08-06:
Closed for null-safety is supported. File iOS support for another issue please

closed 2021-08-06 by Sunbreak