woodemi/quick_usb #14: Linux Desktop platform initialisation error launching debug (QuickUsbPlatform Interface)

digitalkirkley, 2021-03-17:
Using "develop" branch (which includes updated FFI 1.0.0) within Flutter Desktop Linux, the following error occurs:

Launching lib/main.dart on Linux in debug mode...
ERROR: ../../../../.pub-cache/git/quick_usb-9db68d8c68560526baacc373f9db87f37dd7efcf/lib/src/quick_usb_platform_interface.dart:11:27: Error: Field '_instance' should be initialized because its type 'QuickUsbPlatform' doesn't allow null.
ERROR:  - 'QuickUsbPlatform' is from 'package:quick_usb/src/quick_usb_platform_interface.dart' ('../../../../.pub-cache/git/quick_usb-9db68d8c68560526baacc373f9db87f37dd7efcf/lib/src/quick_usb_platform_interface.dart').
ERROR:   static QuickUsbPlatform _instance;
ERROR:                           ^^^^^^^^^
ERROR: ../../../../.pub-cache/git/quick_usb-9db68d8c68560526baacc373f9db87f37dd7efcf/lib/src/quick_usb_android.dart:24:9: Error: A value of type 'List<Map<dynamic, dynamic>>?' can't be assigned to a variable of type 'List<Map<dynamic, dynamic>>' because 'List<Map<dynamic, dynamic>>?' is nullable and 'List<Map<dynamic, dynamic>>' isn't.
ERROR:  - 'List' is from 'dart:core'.
ERROR:  - 'Map' is from 'dart:core'.
ERROR:         await _channel.invokeListMethod('getDeviceList');
ERROR:         ^
ERROR: ../../../../.pub-cache/git/quick_usb-9db68d8c68560526baacc373f9db87f37dd7efcf/lib/src/quick_usb_android.dart:30:21: Error: A value of type 'Future<dynamic>' can't be returned from a function with return type 'Future<bool>'.
ERROR:  - 'Future' is from 'dart:async'.
ERROR:     return _channel.invokeMethod('hasPermission', usbDevice.toMap());
ERROR:                     ^
ERROR: ../../../../.pub-cache/git/quick_usb-9db68d8c68560526baacc373f9db87f37dd7efcf/lib/src/quick_usb_android.dart:40:21: Error: A value of type 'Future<dynamic>' can't be returned from a function with return type 'Future<bool>'.
ERROR:  - 'Future' is from 'dart:async'.
ERROR:     return _channel.invokeMethod('openDevice', usbDevice.toMap());

ERROR:                     ^
ERROR: ../../../../.pub-cache/git/quick_usb-9db68d8c68560526baacc373f9db87f37dd7efcf/lib/src/quick_usb_android.dart:58:21: Error: A value of type 'Future<dynamic>' can't be returned from a function with return type 'Future<bool>'.
ERROR:  - 'Future' is from 'dart:async'.
ERROR:     return _channel.invokeMethod('setConfiguration', config.toMap());
ERROR:                     ^
ERROR: ../../../../.pub-cache/git/quick_usb-9db68d8c68560526baacc373f9db87f37dd7efcf/lib/src/quick_usb_android.dart:63:21: Error: A value of type 'Future<dynamic>' can't be returned from a function with return type 'Future<bool>'.
ERROR:  - 'Future' is from 'dart:async'.
ERROR:     return _channel.invokeMethod('claimInterface', intf.toMap());
ERROR:                     ^
ERROR: ../../../../.pub-cache/git/quick_usb-9db68d8c68560526baacc373f9db87f37dd7efcf/lib/src/quick_usb_android.dart:68:21: Error: A value of type 'Future<dynamic>' can't be returned from a function with return type 'Future<bool>'.
ERROR:  - 'Future' is from 'dart:async'.
ERROR:     return _channel.invokeMethod('releaseInterface', intf.toMap());
ERROR:                     ^
ERROR: ../../../../.pub-cache/git/quick_usb-9db68d8c68560526baacc373f9db87f37dd7efcf/lib/src/quick_usb_android.dart:88:21: Error: A value of type 'Future<dynamic>' can't be returned from a function with return type 'Future<int>'.
ERROR:  - 'Future' is from 'dart:async'.
ERROR:     return _channel.invokeMethod('bulkTransferOut', {
ERROR:                     ^

Exception: Build process failed
Exited (sigterm)

Sunbreak, 2021-03-17:
Maybe related to https://github.com/woodemi/quick_usb/blob/48c6f5e6038178295577f64ae9a0c8e84f1e874a/lib/src/quick_usb_desktop.dart#L25-L31

Could you try different path for `Libusb(DynamicLibrary.open('lib/libusb-1.0.23.so'))`?

digitalkirkley, 2021-03-17:
> Maybe related to
> 
> https://github.com/woodemi/quick_usb/blob/48c6f5e6038178295577f64ae9a0c8e84f1e874a/lib/src/quick_usb_desktop.dart#L25-L31
> 
> Could you try different path for `Libusb(DynamicLibrary.open('lib/libusb-1.0.23.so'))`?

Unfortunately, still getting the same: 
[   +6 ms] [+25032 ms] ../quick_usb-develop/lib/src/quick_usb_platform_interface.dart:11:27: Error: Field '_instance' should be
initialized because its type 'QuickUsbPlatform' doesn't allow null.

digitalkirkley, 2021-03-17:
To resolve - I think maybe making some of the declarations nullable E.g. QuickUsbPlatform**?** _instance; and doing same for the other variables also.

digitalkirkley, 2021-03-17:
I have updated the various variables and resolved the errors for null safety.  I have also went back to the original path but have made an adjustment as its './build/linux/arm64/debug/bundle/lib/libusb-1.0.23.so'.  However, I realise this is for Linux X86 and I am testing on arm64.  I will try this on an Intel X64 device.  In meantime, is it easy to swap in the libusb arm64?  (apologies, I ain't Linux savvy here).

Sunbreak, 2021-03-17:
Sorry, I am not quite familiar too

Sunbreak, 2021-03-18:
> To resolve - I think maybe making some of the declarations nullable E.g. QuickUsbPlatform**?** _instance; and doing same for the other variables also.

PRs are welcome

digitalkirkley, 2021-03-18:
Super, thank you.  I intend to PR all of this back very soon - needed to prove working on X86 and it does.  I believe it might be the native .so file for Libusb not being arm64.  We can resolve this later.  Appreciate help and will do a PR.

Sunbreak, 2021-03-18:
Does Flutter Linux support x86 or arm64?

As far as I known, Flutter Linux works on x64. Flutter Windows only works on x64 for right now

Sunbreak, 2021-03-30:
Accoding to https://medium.com/flutter-community/build-and-deploy-native-c-libraries-with-flutter-cc7531d590b5:

> # Linux
> The steps for integrating an external library on Linux are otherwise the same as on Windows, except that locating the library at runtime requires some additional steps. Here’s an example that utilizes the Flutter plugin code to resolve the correct path, and uses an environment variable to pass the information to Dart:

There seems to be a workaround: https://gist.github.com/jpnurmi/4d14b1ef055d8635fb8e69dd1f9d90a7

```diff
diff --git a/linux/CMakeLists.txt b/linux/CMakeLists.txt
index 0bac83e..6ec6ea1 100644
--- a/linux/CMakeLists.txt
+++ b/linux/CMakeLists.txt
@@ -18,6 +18,8 @@ target_link_libraries(${PLUGIN_NAME} PRIVATE PkgConfig::GTK)
 
 # List of absolute paths to libraries that should be bundled with the plugin
 set(flutter_foo_plugin_bundled_libraries
-  ""
+  "$<TARGET_FILE:foo>"
   PARENT_SCOPE
 )
+
+add_subdirectory(path/to/foo)
diff --git a/linux/flutter_foo_plugin_plugin.cc b/linux/flutter_foo_plugin_plugin.cc
index 03a6b5f..c490270 100644
--- a/linux/flutter_foo_plugin_plugin.cc
+++ b/linux/flutter_foo_plugin_plugin.cc
@@ -3,6 +3,7 @@
 #include <flutter_linux/flutter_linux.h>
 #include <gtk/gtk.h>
 #include <sys/utsname.h>
+#include <glib.h>
 
 #define FLUTTER_FOO_PLUGIN(obj) \
   (G_TYPE_CHECK_INSTANCE_CAST((obj), flutter_foo_plugin_get_type(), \
@@ -51,10 +52,33 @@ static void method_call_cb(FlMethodChannel* channel, FlMethodCall* method_call,
   flutter_foo_plugin_handle_method_call(plugin, method_call);
 }
 
+// Gets the directory the current executable is in, borrowed from:
+// https://github.com/flutter/engine/blob/master/shell/platform/linux/fl_dart_project.cc#L27
+//
+// Copyright 2013 The Flutter Authors. All rights reserved.
+// Use of this source code is governed by a BSD-style license that can be
+// found in https://github.com/flutter/engine/blob/master/LICENSE.
+static gchar* get_executable_dir() {
+  g_autoptr(GError) error = nullptr;
+  g_autofree gchar* exe_path = g_file_read_link("/proc/self/exe", &error);
+  if (exe_path == nullptr) {
+    g_critical("Failed to determine location of executable: %s",
+               error->message);
+    return nullptr;
+  }
+
+  return g_path_get_dirname(exe_path);
+}
+
 void flutter_foo_plugin_register_with_registrar(FlPluginRegistrar* registrar) {
   FlutterFooPlugin* plugin = FLUTTER_FOO_PLUGIN(
       g_object_new(flutter_foo_plugin_get_type(), nullptr));
 
+  g_autofree gchar* executable_dir = get_executable_dir();
+  g_autofree gchar* libfoo_path =
+      g_build_filename(self_exe, "lib", "libfoo.so", nullptr);
+  setenv("LIBFOO_PATH", libfoo_path, 0);
+
   g_autoptr(FlStandardMethodCodec) codec = fl_standard_method_codec_new();
   g_autoptr(FlMethodChannel) channel =
       fl_method_channel_new(fl_plugin_registrar_get_messenger(registrar),
```

Sunbreak, 2021-08-06:
Fixed by https://github.com/woodemi/quick_usb/pull/22

closed 2021-08-06 by Sunbreak