woodemi/quick_usb #15: Browse the device

agnoam, 2021-04-08:
Hey,
I want to get the USB path, for browsing the device through my application.
How do I get this done ?

Sunbreak, 2021-04-08:
`USB path` has different meaning on different OS. You'd better check the implementation of https://github.com/woodemi/libusb.dart

agnoam, 2021-04-08:
This plugin is for desktop (windows, linux and macos)
I'm searching one for Android

Sunbreak, 2021-04-08:
`quick_usb` is mainly to provide cross-platform access to USB devices.

You could wrap your own plugin with [Android UsbManager](https://developer.android.com/reference/android/hardware/usb/UsbManager)

shaxxx, 2021-10-14:
> This plugin is for desktop (windows, linux and macos) I'm searching one for Android

This plugin supports Android and uses Android UsbManager under the hood which is again Android wrapper for libusb.
It's unclear what you mean by "browsing" but if you're reffering to a path of USB disk devices to be browsed by some kind of file explorer than you'll either need to do some extra work by yourself or look in another place.

Please provide more info about your application and what you want to do.

Sunbreak, 2022-02-04:
I'd like to close it if no further info

closed 2022-02-04 by Sunbreak