woodemi/quick_usb #20: How to send POS print data

hamzamon, 2021-05-28:
How to send POS / ESC print commands to the printer device, can you give me an example ?

Sunbreak, 2021-05-28:
Does your POS / ESC have any Android SDK or Example? Usually decompiling java helps a lot

hamzamon, 2021-05-28:
I am using Bixolon USB POS printer, but I could not find sdk to use in flutter. Can I send print command using quick_usb plugin ?

hamzamon, 2021-05-28:
I mean in desktop not android

shaxxx, 2021-08-18:
It doesn't matter desktop or android, principle is same.
You don't need any printer SDK or decompiling if your printer supports ESC/POS (i haven't seen one without it).
Using example app

1.  init
2.  getDeviceList
3.  openDevice (make sure you specify your POS device, example always uses first device, which might not be what you need.)
3.  getConfiguration
4. bulkTransferOut

For number 5 to actually print something find line 

`var data = Uint8List.fromList(utf8.encode(''));`

which doesn't actually sends anything, and replace it with this
```

var posBytes = [
              // init printer
              0x1B, // Esc
              0x40, // @
              // select character code table
              0x1B, // Esc
              0x74, // t
              0x8, // 8  <-- location of CodePage 852 in my printer, usually 0x8 or 0x12
              0x54, // T
              0x45, // E
              0x53, // S
              0x54, // T
              0x20, // Space
              0xE6, // Š
              0xD1, // Đ
              0xAC, // Č
              0x8F, // Ć
              0xA6, // Ž
              0xE7, // š
              0xD0, // đ
              0x9F, // č
              0x86, // ć
              0xA7, // ž
              //add new line
              0x0D, // CR (carriage return)
              0x0A, // NL (new line)
              //add new line
              0x0D, // CR (carriage return)
              0x0A, // NL (new line)
              //add new line
              0x0D, // CR (carriage return)
              0x0A, // NL (new line)
              //add new line
              0x0D, // CR (carriage return)
              0x0A, // NL (new line)
              // init printer
              0x1B, // Esc
              0x40, // @
            ];
            await QuickUsb.detachKernelDriver(_configuration.interfaces[0]);
            var data = Uint8List.fromList(posBytes);
```

If you printer has code page 852 stored at location 0x8 in printer memory you should see

`TEST ŠĐČĆŽšđčćž`

There are plenty of libraries on pub.dev that allow you to convert commands to sequence of bytes.

hamzamon, 2021-08-18:
Thank you a lot, let me try and will update.

s183898, 2021-09-11:
Any update on this?

shaxxx, 2021-09-11:
What kind of update? Have you tried the to run the example?

s183898, 2021-09-11:
Yes, but without any luck. I am attempting to use a thermal printer from and android tablet using a USB connection.

shaxxx, 2021-09-11:
If I understand correctly, you followed my instructions from above and printer did not print anything?
If that is the case than either your printer does not support ESC/POS commands (highly unlikely if it's POS receipt printer), or you misconfigured something.
Double check if you followed my instructions and selected right port/interface in the code (example always uses first).
Bottom line this library already supports sending bytes to POS printer, there's nothing more to be done.
And it works so on Linux, MacOS and Android (haven't tested it on Windows since it's pain in the ass).
Library itself just sends bytes you provided to interface/configuration you select. It doesn't implement ESC/POS commands, find a lib on pub.dev that implements ESC/POS commands, use that same lib to convert those commands to bytes and then use this lib to transfer bytes to usb printer.

s183898, 2021-09-11:
My bad, it had nothing to do with printing, but simply my pubspec.yaml file which did not match the package.

s183898, 2021-09-11:
Thanks for the elaborate answer by the way! I tried following the steps, and the printer appears on the app (as the only device), and the message "bulkTransferOut 30" appears when I click , but the printer does not print.

fanchou, 2021-09-13:
thanks, it works for me. 
my steps:

init ---> getDevicesWithDescription --> select device ---> openDevice---> getConfiguration--->**claimInterface**---> bulkTransferOut   

I use [esc_pos_utils](https://github.com/andrey-ushakov/esc_pos_utils) for the base ESC/POS command package.

s183898, 2021-09-13:
nice to hear fanchou. Could you by any chance share an example combining the esc_pos_utils with the quick_usb packages?

Sunbreak, 2021-09-13:
For anyone interested in Flutter Desktop: https://github.com/leanflutter/awesome-flutter-desktop

WeChat: wk2311007 （for @fanchou 周凡）
Discord: https://discord.gg/vba8W9SF

fanchou, 2021-09-15:
> nice to hear fanchou. Could you by any chance share an example combining the esc_pos_utils with the quick_usb packages?

https://github.com/fanchou/flutter_escpos       just test on Mac now

shaxxx, 2021-09-21:
> How to send byte data to the printer in windows ?

This issue is resolved and should be closed 🔒
Plugin supports POS printers ad well as any USB device, and example has been provided.
Please keep in mind this is not general forum and discussion should be plugin specific.

closed 2021-09-22 by Sunbreak