woodemi/quick_usb #23: Not working on MacOS

MateoSegura, 2021-07-05:
Hi, I'm running MacOS 10.15, stable channel of latest flutter & dart.

No errors on my flutter doctor.

I get the following when I try to run the app.

I tried to look for my libusb location, and it's different from what the warning says. 

How can I change the path where it looks for the library? Or what should I do in order to fix this?

[image]

Sunbreak, 2021-07-06:
Sorry, it's an `rpath` mistake in the `libusb.dylib`. You could `brew install libusb` to work around for now.

I'll update the `rpath` in the dylib later

MateoSegura, 2021-07-06:
I had already installed libusb with that command. 

However, the library version is 1.0.24.

The library is there, it's just a different path.

Sunbreak, 2021-07-06:
According to https://github.com/Sunbreak/cronet.dart#macos-2, you could change the `rpath` yourself

```md
**CAUTION**

`id` may not be compatible with your detination application/library

```sh
$ otool -D libcronet.86.0.4240.198.dylib
cronet/macos/libcronet.86.0.4240.198.dylib:
@executable_path/libcronet.86.0.4240.198.dylib
```

Change `id` with install_name_tool

```sh
install_name_tool -id @rpath/libcronet.86.0.4240.198.dylib libcronet.86.0.4240.198.dylib
```
```

closed 2021-07-06 by Sunbreak

MateoSegura, 2021-07-06:
Even when I change the path it still displays the same error.

I check both folders library is installed.

Do you think you could possibly update the rpath? I really don't know what to do to fix it...

Sunbreak, 2021-07-06:
Did #24 fix it?

MateoSegura, 2021-07-06:
It did fix it, but only for the example. If I try to implement quick usb in a different project it gives me the same error. 

what files//libraries do I need to change in my project to make it work.

Also, thank you for helping me out!

Sunbreak, 2021-07-06:
> It did fix it, but only for the example. If I try to implement quick usb in a different project it gives me the same error.

What do you mean `implement quick usb`?

MateoSegura, 2021-07-06:
I mean I have a project that displays a simple gauge. This gauge displays the value from USB port using an Arduino. I want to use the quick_usb package to be able to read this information in my app, as well as send small commands to the Arduino. 

However, If I try to add the dependency to my pubspec.yml, the same error happens. 

I figure a work around was to use the path of the library I downloaded from GitHub, and it works no problem.

Sunbreak, 2021-07-06:
Yep. Depending on `pub.dev` versin won't work because I didn't publish the fixed version.

I'll find my time later

PR #24 (2021-07-06): Fix dylib path
fix #23