woodemi/quick_usb #25: How to properly read data with this library?

MateoSegura, 2021-07-06:
Hello,

I am trying to use this library with an Arduino to create a simple data logger. The problem is that the Arduino constantly outputs data & I need update a list on my flutter program with such data. How can I implement something like this with the bulk transfer method? I do not understand what the end point is, it says that device has endpoint '1'. Any ideas or suggestions are greatly appreciated.

[image]

Sunbreak, 2021-07-06:
Sorry. The lib is mostly a wrapper of libusb: https://github.com/libusb/libusb on Flutter PC. You'd better learn for libusb

MateoSegura, 2021-07-06:
Ok, thank you!. I'll look into it

closed 2021-07-06 by Sunbreak