woodemi/quick_usb #26: Issue on Linux

mjafartp, 2021-07-30:
Hi,
it's not working on Linux 
on setConfig giving an error 

> setConfig error: LIBUSB_ERROR_BUSY

Sunbreak, 2021-07-30:
Seem libusb is busy. I'm not quite familiar with Linux. Sorry

mjafartp, 2021-07-31:
@Sunbreak this not only for a single device, this error we getting for all devices.

I think the issue with `claimInterface` on this is we always getting a false response

Sunbreak, 2021-08-01:
Have you tried libusb with C program? I used to debug problems with C/C++

mjafartp, 2021-08-01:
Can you please share a C Code to send data to USB, I can check from my side. 
my Device detail `Vendor:Device = 0483:5743`

mjafartp, 2021-08-01:
hi,

I have found that the issue with `libusb_detach_kernel_driver ( libusb_device_handle *dev,int interface)` in Linux needs to detach it from the driver first to claim interface.

shaxxx, 2021-08-18:
Instead of explicitly detaching device from kernel driver permanently which can cause problems for other applications using same device over kernel driver [libusb_set_auto_detach_kernel_driver()](https://libusb.sourceforge.io/api-1.0/group__libusb__dev.html#gac35b26fef01271eba65c60b2b3ce1cbf) should be preffered as it automatically detaches/attaches device when needed.