woodemi/quick_usb #38: Not getting Manufacturer, product and Serial Number on Windows

utpalruparel, 2021-10-04:
Hello, First of all thanks for the brilliant library. It works for me on MacOS. Now I am trying to use same device in windows 10 but it does not find the attached device. Any extra settings we need to do? I am getting null for all Manugacturer, product and serial number for windows. 

Let me know.

Thanks

Sunbreak, 2021-10-08:
Which API do you call? Could you post a sample code?

Sunbreak, 2021-10-11:
Could you `flutter pub run listdevs.dart` under  https://github.com/woodemi/libusb.dart and post the result here?

utpalruparel, 2021-10-14:
I am using sample code.. It shows connected USB devices in MacOS. But same it does not show any in Windows.

utpalruparel, 2021-10-14:
> `flutter pub run example/listdevs.dart` under [https://github.com/woodemi/libusb.dar](https://github.com/woodemi/libusb.dart)

I Am not able to Run this command. It shows me errors

shaxxx, 2021-10-14:
You did not answer my question, but regardless, this output does not mean Quick USB is not working.
See my example from Win10, there's plenty of USB devices (including POS Printer that does show description on other platforms but not on Windows) but only first USB device has actual description.
[image]
Windows automatically uses it's own USB subsystem when devices is attached.

In your scenario I believe Quick USB is not the answer (at least not on Windows).
I can only recommend you what I use on Windows for POS printing
https://github.com/shaxxx/windows_raw_print
It basically sends raw bytes to printer installed in windows (use Generic Text driver).
DISCLAIMER: I'm the author.

shaxxx, 2021-10-14:
There are at least three solution

- use Quick USB AS IS, if you always use same printer you can use device/manufacturer ID to recognize printer, no need for description
- use Zadig or your own in house installer to replace default WinUsb driver on your client machines
- use my plugin

With all these said, it's clear that this is not issue with Quick Usb and this issue should be closed.

closed 2021-10-14 by Sunbreak