woodemi/quick_usb #42: build web issue

selvam920, 2021-11-29:
i'm using this plugin only on android when i'm building the web i'm getting below error.

b-0.3.23-nullsafety.0/lib/libusb64.dart:856:9:
Error: 'Pointer' isn't a type.
    ffi.Pointer<libusb_transfer> transfer,
        ^^^^^^^
/C:/Users/Jack%20Selvam/AppData/Local/Pub/Cache/hosted/pub.dartlang.org/libus
b-0.3.23-nullsafety.0/lib/libusb64.dart:868:9:
Error: 'Pointer' isn't a type.
    ffi.Pointer<libusb_transfer> transfer,
        ^^^^^^^
/C:/Users/Jack%20Selvam/AppData/Local/Pub/Cache/hosted/pub.dartlang.org/libus
b-0.3.23-nullsafety.0/lib/libusb64.dart:882:9:
Error: 'Pointer' isn't a type.
    ffi.Pointer<libusb_transfer> transfer,
        ^^^^^^^
/C:/Users/Jack%20Selvam/AppData/Local/Pub/Cache/hosted/pub.dartlang.org/libus
b-0.3.23-nullsafety.0/lib/libusb64.dart:894:9:
Error: 'Pointer' isn't a type.
    ffi.Pointer<libusb_device_handle> dev_handle,
        ^^^^^^^
/C:/Users/Jack%20Selvam/AppData/Local/Pub/Cache/hosted/pub.dartlang.org/libus
b-0.3.23-nullsafety.0/lib/libusb64.dart:899:21:
Error: 'Uint8' isn't a type.
    ffi.Pointer<ffi.Uint8> data,
                    ^^^^^
/C:/Users/Jack%20Selvam/AppData/Local/Pub/Cache/hosted/pub.dartlang.org/libus
b-0.3.23-nullsafety.0/lib/libusb64.dart:899:9:
Error: 'Pointer' isn't a type.
    ffi.Pointer<ffi.Uint8> data,
        ^^^^^^^
/C:/Users/Jack%20Selvam/AppData/Local/Pub/Cache/hosted/pub.dartlang.org/libus
b-0.3.23-nullsafety.0/lib/libusb64.dart:920:9:
Error: 'Pointer' isn't a type.
    ffi.Pointer<libusb_device_handle> dev_handle,
        ^^^^^^^
/C:/Users/Jack%20Selvam/AppData/Local/Pub/Cache/hosted/pub.dartlang.org/libus
b-0.3.23-nullsafety.0/lib/libusb64.dart:922:21:
Error: 'Uint8' isn't a type.
    ffi.Pointer<ffi.Uint8> data,
                    ^^^^^
/C:/Users/Jack%20Selvam/AppData/Local/Pub/Cache/hosted/pub.dartlang.org/libus
b-0.3.23-nullsafety.0/lib/libusb64.dart:922:9:
Error: 'Pointer' isn't a type.
    ffi.Pointer<ffi.Uint8> data,
        ^^^^^^^
/C:/Users/Jack%20Selvam/AppData/Local/Pub/Cache/hosted/pub.dartlang.org/libus
b-0.3.23-nullsafety.0/lib/libusb64.dart:924:21:
Error: 'Int32' isn't a type.
    ffi.Pointer<ffi.Int32> actual_length,
                    ^^^^^
/C:/Users/Jack%20Selvam/AppData/Local/Pub/Cache/hosted/pub.dartlang.org/libus
b-0.3.23-nullsafety.0/lib/libusb64.dart:924:9:
Error: 'Pointer' isn't a type.
    ffi.Pointer<ffi.Int32> actual_length,
        ^^^^^^^
/C:/Users/Jack%20Selvam/AppData/Local/Pub/Cache/hosted/pub.dartlang.org/libus
b-0.3.23-nullsafety.0/lib/libusb64.dart:942:9:
Error: 'Pointer' isn't a type.
    ffi.Pointer<libusb_device_handle> dev_handle,
        ^^^^^^^
/C:/Users/Jack%20Selvam/AppData/Local/Pub/Cache/hosted/pub.dartlang.org/libus
b-0.3.23-nullsafety.0/lib/libusb64.dart:944:21:
Error: 'Uint8' isn't a type.
    ffi.Pointer<ffi.Uint8> data,
                    ^^^^^
/C:/Users/Jack%20Selvam/AppData/Local/Pub/Cache/hosted/pub.dartlang.org/libus
b-0.3.23-nullsafety.0/lib/libusb64.dart:944:9:
Error: 'Pointer' isn't a type.
    ffi.Pointer<ffi.Uint8> data,
[... 909 lines cut ...]
b-0.3.23-nullsafety.0/lib/libusb64.dart:2235:16:
Error: Only JS interop members may be 'external'.
  external int status;
               ^
/C:/Users/Jack%20Selvam/AppData/Local/Pub/Cache/hosted/pub.dartlang.org/libus
b-0.3.23-nullsafety.0/lib/libusb64.dart:2295:16:
Error: Only JS interop members may be 'external'.
  external int tv_sec;
               ^
/C:/Users/Jack%20Selvam/AppData/Local/Pub/Cache/hosted/pub.dartlang.org/libus
b-0.3.23-nullsafety.0/lib/libusb64.dart:2298:16:
Error: Only JS interop members may be 'external'.
  external int tv_usec;
               ^
/C:/Users/Jack%20Selvam/AppData/Local/Pub/Cache/hosted/pub.dartlang.org/libus
b-0.3.23-nullsafety.0/lib/libusb64.dart:2306:16:
Error: Only JS interop members may be 'external'.
  external int fd;
               ^
/C:/Users/Jack%20Selvam/AppData/Local/Pub/Cache/hosted/pub.dartlang.org/libus
b-0.3.23-nullsafety.0/lib/libusb64.dart:2313:16:
Error: Only JS interop members may be 'external'.
  external int events;
               ^
Error: Compilation failed.

Sunbreak, 2021-11-29:
Sorry it doen't support web yet

closed 2021-11-29 by Sunbreak

selvam920, 2021-11-29:
i dont want support web, but app should be build

Sunbreak, 2021-11-30:
Then you should use conditional import
- https://medium.com/@dvargahali/dart-2-conditional-imports-update-16147a776aa8
- https://dart.dev/guides/libraries/create-library-packages#conditionally-importing-and-exporting-library-files

selvam920, 2021-11-30:
can it be from this plugin? bcoz many people to download this plugin they will face this issue

Sunbreak, 2021-11-30:
I don't think so. Flutter Web or AugularDart are far from popular

Actually I worked on some projects on Flutter Web last year. I don't think it is a good idea to use it now.

Especially on some project depend on USB

---

If you're interested, you could checkout our other projects and learn how to make it work on Web

- https://github.com/woodemi/quick_blue
- https://github.com/woodemi/notepad_core
- https://github.com/woodemi/quick-scan-flutter

jackselvam, 2021-11-30:
I just saying conditional import from plugin side

Sunbreak, 2021-11-30:
> I just saying conditional import from plugin side

Conditional import work only with `dart.library.io` and `dart.library.js`

If `quick_usb` use conditional import, there is only an empty implementation left -- because ALL of `quick_usb` functions are base on `dart.library.io`

> bcoz many people to download this plugin they will face this issue

I don't think Flutter Web or AugularDart is popular enough that `many` people would face this issue

Sunbreak, 2021-11-30:
If you are interested, you could try [WebUSB](https://wicg.github.io/webusb/)

I was planning to implement `quick_usb_web` via Chrome WebUSB API https://web.dev/usb/

That is another story though

selvam920, 2021-11-30:
The below plugin we can not use web but when we build the project for web it should.
like that i'm asking,
you need to include platform interface
https://github.com/bitsdojo/bitsdojo_window

Sunbreak, 2021-11-30:
Yep. `quick_usb_platform_interface` would help.

But it is quite cumbersome to develop and maintain. Checkout change logs and tags of [quick_blue](https://github.com/woodemi/quick_blue) and [notepad_core](https://github.com/woodemi/notepad_core)

`quick_*` series means quick to understand, learn, use and maintain

---

Forks and PRs are welcome