woodemi/quick_usb #45: libusb 1.0.24

redDwarf03, 2022-01-29:
hello

could you migrate the project to libusb 1.0.24 in each platform please
thx

Sunbreak, 2022-02-04:
Why do we need libusb 1.0.24?

fanchou, 2022-02-05:
v1.0.25 is release.

Sunbreak, 2022-02-06:
Is there any benefit to upgrade? More APIs? More reliable? More compatiliblity?

Is there any clue or changelog?

Sunbreak, 2022-02-06:
Upgrade to libusb-v1.0.25 requires `re-generation` of https://github.com/woodemi/libusb.dart, i.e. `flutter run ffigen`

**`ffi` and `ffigen` are quite different to one year ago**

There are `ffi-1.1.2` and `ffigen-2.4.2` inside [`libusb.dart`'s lock file](https://github.com/woodemi/libusb.dart/blob/ba945791b21786f25392d96a57ce926eb2295eef/pubspec.lock#L81-L94)

```yaml
  ffi:
    dependency: "direct main"
    description:
      name: ffi
      url: "https://pub.dartlang.org"
    source: hosted
    version: "1.1.2"
  ffigen:
    dependency: "direct dev"
    description:
      name: ffigen
      url: "https://pub.dartlang.org"
    source: hosted
    version: "2.4.2"
```

Lastest on Pub are https://pub.dev/packages/ffi/versions/1.2.0-dev.0 and https://pub.dev/packages/ffigen/versions/5.0.0-dev.0

PRs are welcome if you're interested

Sunbreak, 2022-02-06:
Closing this in favor of https://github.com/woodemi/libusb.dart/issues/16

closed 2022-02-06 by Sunbreak