woodemi/quick_usb #46: Cannot detect usb connected devices on macOS Monterey

jyashi, 2022-02-02:
`void main() async {
  await QuickUsb.init();
  var deviceList = await QuickUsb.getDeviceList()
      .then((value) => print("debug device list $value"));
 
 
  await GetStorage.init();
  
}`

The print outputs an empty list [] but running flutter devices shows me the correct list of connected devices.

Sunbreak, 2022-02-02:
`QuickUsb.getDeviceList` has nothing to do with `flutter devices`

# `QuickUsb.getDeviceList`

uses `libusb_get_device_list` on desktop: https://libusb.sourceforge.io/api-1.0/group__libusb__dev.html#gac0fe4b65914c5ed036e6cbec61cb0b97

# `flutter devices`

Command | Example of use | More information
--- | --- | ---
devices | `flutter devices -d <DEVICE_ID>` | List all connected devices.

jyashi, 2022-02-02:
I didn't say they had anything to do with each other. What I was trying to
say is that the device is indeed connected with the desktop and there is no
faulty usb cable and yet this plug in command does not return a list of the
connected devices as per the documentation.

jyashi, 2022-02-03:
> Could you post flutter doctor -v?

As requested.

[image: Screen Shot 2022-02-03 at 9.30.09.png]

Best regards,
Navtej Dhot

Sunbreak, 2022-02-03:
I'm not sure if libusb works on ARM64 macOS or connected Android devices

I'll test on my ARM64 macOS as soon as possible

jyashi, 2022-02-03:
Thanks a lot. FYI even though i am testing in macOS right now the app is being developed for Windows desktop exclusively.. So it would be great if it works on Windows 10.

Sunbreak, 2022-02-03:
`QuickUsb.getDeviceList` works on x64 macOS 11.2.3 with Pixel 4

> deviceList [{identifier: 25, vendorId: 6353, productId: 20199, configurationCount: 1}]

According to http://www.linux-usb.org/usb.ids, It is VID-`18d1`(Google) PID-`4ee7`(Pixel)

<details>
<summary>`flutter doctor -v`</summary>

```sh
[✓] Flutter (Channel stable, 2.0.6, on macOS 11.2.3 20D91 darwin-x64, locale en-CN)
    • Flutter version 2.0.6 at /Users/sunbreak/flutter/flutter-2.0.x
    • Framework revision 1d9032c7e1 (9 months ago), 2021-04-29 17:37:58 -0700
    • Engine revision 05e680e202
    • Dart version 2.12.3
    • Pub download mirror https://pub.flutter-io.cn
    • Flutter download mirror https://storage.flutter-io.cn

[✓] Android toolchain - develop for Android devices (Android SDK version 30.0.3)
    • Android SDK at /Users/sunbreak/Library/Android/sdk
    • Platform android-30, build-tools 30.0.3
    • Java binary at: /Library/Java/JavaVirtualMachines/jdk1.8.0_281.jdk/Contents/Home/bin/java
    • Java version Java(TM) SE Runtime Environment (build 1.8.0_281-b09)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 12.2, Build version 12B45b
    • CocoaPods version 1.10.1

[✓] Chrome - develop for the web
    • Chrome at /Applications/Google Chrome.app/Contents/MacOS/Google Chrome

[!] Android Studio (version 2020.3)
    • Android Studio at /Users/sunbreak/Library/Application Support/JetBrains/Toolbox/apps/AndroidStudio/ch-0/203.7678000/Android Studio.app/Contents
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    ✗ Unable to find bundled Java version.
    • Try updating or re-installing Android Studio.

[✓] VS Code (version 1.63.2)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.29.0

[✓] Connected device (3 available)
    • Pixel 4 (mobile) • 9A261FFAZ0040M • android-arm64  • Android 12 (API 31)
    • macOS (desktop)  • macos          • darwin-x64     • macOS 11.2.3 20D91 darwin-x64
    • Chrome (web)     • chrome         • web-javascript • Google Chrome 98.0.4758.80

! Doctor found issues in 1 category.
```
</details>

Sunbreak, 2022-02-04:
`QuickUsb.getDeviceList` works on ARM64 macOS 11.6.3 with Pixel 4

> deviceList [{identifier: 1, vendorId: 6353, productId: 20199, configurationCount: 1}]

<details>
<summary>`flutter doctor -v`</summary>

```sh
[✓] Flutter (Channel stable, 2.8.1, on macOS 11.6.3 20G415 darwin-arm, locale zh-Hans-CN)
    • Flutter version 2.8.1 at /Users/pyy/google/flutter-2.8.x
    • Upstream repository https://github.com/flutter/flutter
    • Framework revision 77d935af4d (7 weeks ago), 2021-12-16 08:37:33 -0800
    • Engine revision 890a5fca2e
    • Dart version 2.15.1
    • Pub download mirror https://pub.flutter-io.cn
    • Flutter download mirror https://storage.flutter-io.cn

[✓] Android toolchain - develop for Android devices (Android SDK version 32.1.0-rc1)
    • Android SDK at /Users/pyy/Library/Android/sdk
    • Platform android-32, build-tools 32.1.0-rc1
    • Java binary at: /Applications/Android Studio.app/Contents/jre/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 11.0.11+0-b60-7772763)
    • All Android licenses accepted.

[!] Xcode - develop for iOS and macOS (Xcode 12.5.1)
    • Xcode at /Applications/Xcode-12.5.1.app/Contents/Developer
    ! Flutter recommends a minimum Xcode version of 13.0.0.
      Download the latest version or update via the Mac App Store.
    • CocoaPods version 1.11.2

[✓] Chrome - develop for the web
    • Chrome at /Applications/Google Chrome.app/Contents/MacOS/Google Chrome

[✓] Android Studio (version 2021.1)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.11+0-b60-7772763)

[✓] VS Code (version 1.64.0)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.34.0

[!] Proxy Configuration
    • HTTP_PROXY is set
    ! NO_PROXY is not set

[✓] Connected device (3 available)
    • Pixel 4 (mobile) • 9A261FFAZ0040M • android-arm64  • Android 12 (API 31)
    • macOS (desktop)  • macos          • darwin-arm64   • macOS 11.6.3 20G415 darwin-arm
    • Chrome (web)     • chrome         • web-javascript • Google Chrome 98.0.4758.80

! Doctor found issues in 2 categories.
```
</details>

Sunbreak, 2022-02-04:
@jyashi Could you try run [libusb](https://github.com/woodemi/libusb.dart) example?

```sh
% git clone https://github.com/woodemi/libusb.dart -b develop-3.23-nullsafety
% cd libusb.dart
% flutter pub get
% flutter pub run example/listdevs.dart
18d1:4ee7 (bus 0, device 1)
 path: 01
```
PS: `18d1:4ee7` is `VID:PID`, i.e. Pixel 4

jyashi, 2022-02-04:
Ran as instructed..
[image]

Sunbreak, 2022-02-04:
Then `QuickUsb.getDeviceList` should work on your environment

jyashi, 2022-02-04:
> Then `QuickUsb.getDeviceList` should work on your environment

Thanks for your support.

By the way i think this plugin would have a great adoption if you have any YouTube videos showing Flutter UI controlling esp32 or M5stack microcontrollers via usb serial communication using this plugin.

Sunbreak, 2022-02-04:
Sorry. I dont have any of esp32 or M5stack

closed 2022-02-04 by Sunbreak