woodemi/quick_usb #48: not working on macos monterey

Bikram40, 2022-02-25:
getting empty device list

Sunbreak, 2022-02-25:
Paste `flutter doctor -v` please

Bikram40, 2022-02-26:
[✓] Flutter (Channel stable, 2.10.1, on macOS 12.1 21C52 darwin-x64, locale en-IN)
    • Flutter version 2.10.1 at /Users/bikrammeher/Downloads/flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision db747aa133 (2 weeks ago), 2022-02-09 13:57:35 -0600
    • Engine revision ab46186b24
    • Dart version 2.16.1
    • DevTools version 2.9.2

[!] Android toolchain - develop for Android devices (Android SDK version 32.0.0)
    • Android SDK at /Users/bikrammeher/Library/Android/sdk
    • Platform android-31, build-tools 32.0.0
    • ANDROID_HOME = /Users/bikrammeher/Library/Android/sdk
    • Java binary at: /Applications/Android Studio.app/Contents/jre/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 11.0.11+0-b60-7590822)
    ✗ Android license status unknown.
      Run `flutter doctor --android-licenses` to accept the SDK licenses.
      See https://flutter.dev/docs/get-started/install/macos#android-setup for more details.

[✓] Xcode - develop for iOS and macOS (Xcode 13.2)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • CocoaPods version 1.11.2

[✓] Chrome - develop for the web
    • Chrome at /Applications/Google Chrome.app/Contents/MacOS/Google Chrome

[✓] Android Studio (version 2021.1)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.11+0-b60-7590822)

[✓] VS Code (version 1.64.2)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.34.0

[✓] Connected device (2 available)
    • macOS (desktop) • macos  • darwin-x64     • macOS 12.1 21C52 darwin-x64
    • Chrome (web)    • chrome • web-javascript • Google Chrome 98.0.4758.109

[✓] HTTP Host Availability
    • All required HTTP hosts are available

Sunbreak, 2022-02-28:
What about `libusb.dart`'s example?

```
% git clone https://github.com/woodemi/libusb.dart
% cd libusb.dart
% flutter pub get
% flutter pub run example/listdevs.dart
```

Sunbreak, 2022-02-28:
Here is my result on macOS 12.1

- quick_usb list no device by default when nothing connected
- `libusb.dart`'s `example/listdevs.dart` list no device by default when nothing connected

<details>
<summary>flutter doctor -v</summary>

```
[✓] Flutter (Channel stable, 2.10.2, on macOS 12.1 21C52 darwin-arm, locale en-CN)
    • Flutter version 2.10.2 at /Users/wangkun42/google/flutter-2.10.x
    • Upstream repository https://github.com/flutter/flutter
    • Framework revision 097d3313d8 (9 days ago), 2022-02-18 19:33:08 -0600
    • Engine revision a83ed0e5e3
    • Dart version 2.16.1
    • DevTools version 2.9.2
    • Pub download mirror https://pub.flutter-io.cn
    • Flutter download mirror https://storage.flutter-io.cn

[!] Android toolchain - develop for Android devices (Android SDK version 31.0.0)
    • Android SDK at /Users/wangkun42/Library/Android/sdk
    • Platform android-31, build-tools 31.0.0
    • Java binary at: /Users/wangkun42/Library/Application
      Support/JetBrains/Toolbox/apps/AndroidStudio/ch-0/203.7784292/Android
      Studio.app/Contents/jre/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 11.0.10+0-b96-7249189)
    ✗ Android license status unknown.
      Run `flutter doctor --android-licenses` to accept the SDK licenses.
      See https://flutter.dev/docs/get-started/install/macos#android-setup for more details.

[✓] Xcode - develop for iOS and macOS (Xcode 13.2.1)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • CocoaPods version 1.11.2

[✓] Chrome - develop for the web
    • Chrome at /Applications/Google Chrome.app/Contents/MacOS/Google Chrome

[✓] Android Studio (version 2020.3)
    • Android Studio at /Users/wangkun42/Library/Application
      Support/JetBrains/Toolbox/apps/AndroidStudio/ch-0/203.7784292/Android Studio.app/Contents
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.10+0-b96-7249189)

[✓] IntelliJ IDEA Community Edition (version 2021.3.1)
    • IntelliJ at /Users/wangkun42/Library/Application
      Support/JetBrains/Toolbox/apps/IDEA-C/ch-0/213.6461.79/IntelliJ IDEA CE.app
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart

[✓] VS Code (version 1.64.2)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.34.0

[✓] Connected device (1 available)
    • Chrome (web) • chrome • web-javascript • Google Chrome 98.0.4758.109

[✓] HTTP Host Availability
    • All required HTTP hosts are available

! Doctor found issues in 1 category.
```
</details>

Sunbreak, 2022-02-28:
BTW, a lot of HID devices would NOT show in USB devices list on macOS

You could take advantage of https://github.com/Sunbreak/logic_conf.dart/blob/ffbf4b7588be360050881a43a2d70109bf0c324f/bin/k380_conf.dart#L25

```dart
var hidDevices = LogicConf.listDevices()
```

Bikram40, 2022-03-01:
Thank you i got my problem solved, after restarting my mac its working now.

closed 2022-03-01 by Sunbreak