woodemi/quick_usb #50: Cannot run bulkTransferIn on Android properly

serkankavak, 2022-03-04:
Im going to use this package in for an that is running both on Android and Linux. I modified the example app little bit and used it for Android but I have faced with an error that I haven't been able to find any solution.

[modified example app](https://hackmd.io/CDSIV5NfSS-gRwlaK1R72w?both)

From the example app, I go over as 
1.  init
2. getDeviceList
3. hasPermission
4. requestPermission
5. openDevice
6. getConfiguration
7. claimInterface
8. bulkTransferIn

Everything seems alright until I start reading data at step 8. 

The error:
```
E/MethodChannel#quick_usb(26400): Failed to handle method call
E/MethodChannel#quick_usb(26400): java.lang.IllegalArgumentException: Requested element count -1 is less than zero.
E/MethodChannel#quick_usb(26400): 	at kotlin.collections.ArraysKt___ArraysKt.take(_Arrays.kt:4783)
E/MethodChannel#quick_usb(26400): 	at com.example.quick_usb.QuickUsbPlugin.onMethodCall(QuickUsbPlugin.kt:141)
E/MethodChannel#quick_usb(26400): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:262)
E/MethodChannel#quick_usb(26400): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:296)
E/MethodChannel#quick_usb(26400): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$DartMessenger(DartMessenger.java:320)
E/MethodChannel#quick_usb(26400): 	at io.flutter.embedding.engine.dart.-$$Lambda$DartMessenger$TsixYUB5E6FpKhMtCSQVHKE89gQ.run(Unknown Source:12)
E/MethodChannel#quick_usb(26400): 	at android.os.Handler.handleCallback(Handler.java:938)
E/MethodChannel#quick_usb(26400): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/MethodChannel#quick_usb(26400): 	at android.os.Looper.loop(Looper.java:247)
E/MethodChannel#quick_usb(26400): 	at android.app.ActivityThread.main(ActivityThread.java:8676)
E/MethodChannel#quick_usb(26400): 	at java.lang.reflect.Method.invoke(Native Method)
E/MethodChannel#quick_usb(26400): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:602)
E/MethodChannel#quick_usb(26400): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)
E/flutter (26400): [ERROR:flutter/lib/ui/ui_dart_state.cc(209)] Unhandled Exception: PlatformException(error, Requested element count -1 is less than zero., null, java.lang.IllegalArgumentException: Requested element count -1 is less than zero.
E/flutter (26400): 	at kotlin.collections.ArraysKt___ArraysKt.take(_Arrays.kt:4783)
E/flutter (26400): 	at com.example.quick_usb.QuickUsbPlugin.onMethodCall(QuickUsbPlugin.kt:141)
E/flutter (26400): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:262)
E/flutter (26400): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:296)
E/flutter (26400): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$DartMessenger(DartMessenger.java:320)
E/flutter (26400): 	at io.flutter.embedding.engine.dart.-$$Lambda$DartMessenger$TsixYUB5E6FpKhMtCSQVHKE89gQ.run(Unknown Source:12)
E/flutter (26400): 	at android.os.Handler.handleCallback(Handler.java:938)
E/flutter (26400): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/flutter (26400): 	at android.os.Looper.loop(Looper.java:247)
E/flutter (26400): 	at android.app.ActivityThread.main(ActivityThread.java:8676)
E/flutter (26400): 	at java.lang.reflect.Method.invoke(Native Method)
E/flutter (26400): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:602)
E/flutter (26400): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)
E/flutter (26400): )
E/flutter (26400): #0      StandardMethodCodec.decodeEnvelope
E/flutter (26400): #1      MethodChannel._invokeMethod
E/flutter (26400): <asynchronous suspension>
E/flutter (26400): #2      QuickUsbAndroid.bulkTransferIn
E/flutter (26400): <asynchronous suspension>
E/flutter (26400): #3      _MyAppState._bulkTransfer.<anonymous closure>
E/flutter (26400): <asynchronous suspension>
E/flutter (26400):
```

Interestingly, when I use bulkTransferOut, I get the correct output: `I/flutter (26400): bulkTransferOut 2`

Sunbreak, 2022-03-04:
Log please

serkankavak, 2022-03-04:
[√] Flutter (Channel stable, 2.10.2, on Microsoft Windows [Version 10.0.19043.1526], locale en-GB)
    • Flutter version 2.10.2 at C:\Users\Seko\flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision 097d3313d8 (13 days ago), 2022-02-18 19:33:08 -0600
    • Engine revision a83ed0e5e3
    • Dart version 2.16.1
    • DevTools version 2.9.2

[√] Android toolchain - develop for Android devices (Android SDK version 31.0.0)
    • Android SDK at C:\Users\Seko\AppData\Local\Android\sdk
    • Platform android-31, build-tools 31.0.0
    • Java binary at: C:\Program Files\Android\Android Studio\jre\bin\java
    • Java version OpenJDK Runtime Environment (build 11.0.11+9-b60-7590822)
    • All Android licenses accepted.

[√] Chrome - develop for the web
    • Chrome at C:\Program Files\Google\Chrome\Application\chrome.exe

[√] Visual Studio - develop for Windows (Visual Studio Community 2022 17.0.4)
    • Visual Studio at C:\Program Files\Microsoft Visual Studio\2022\Community
    • Visual Studio Community 2022 version 17.0.32014.148
    • Windows 10 SDK version 10.0.19041.0

[√] Android Studio (version 2021.1)
    • Android Studio at C:\Program Files\Android\Android Studio
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.11+9-b60-7590822)

[√] VS Code (version 1.64.2)
    • VS Code at C:\Users\Seko\AppData\Local\Programs\Microsoft VS Code
    • Flutter extension version 3.36.0

[√] Connected device (4 available)
    • SM P610 (mobile)  • 192.168.50.154:5555 • android-arm64  • Android 11 (API 30)
    • Windows (desktop) • windows             • windows-x64    • Microsoft Windows [Version 10.0.19043.1526]
    • Chrome (web)      • chrome              • web-javascript • Google Chrome 98.0.4758.102
    • Edge (web)        • edge                • web-javascript • Microsoft Edge 98.0.1108.62

[√] HTTP Host Availability
    • All required HTTP hosts are available

• No issues found!

Sunbreak, 2022-03-04:
Any logcat on Android device?

serkankavak, 2022-03-04:
Logcat of running the app:

```
List of devices attached
192.168.50.154:5555	device

--------- beginning of system
03-04 17:42:55.716  4957  4995 D SemGameManager: getGMSBinder(), successful: true
03-04 17:42:55.716  4957  4995 D GameManagerService: identifyGamePackage. com.seko.quick_usb_example, mCurrentUserId: 0, callerUserId: 0
03-04 17:42:55.716  4957  4995 D PkgDataHelper: getGamePkgData(). com.seko.quick_usb_example
03-04 17:42:55.716  4957  4957 I CocktailBarManagerServiceContainer: handleMessage: entry what = 5
03-04 17:42:55.719  4957  4995 D GameManagerService: noteResumeComponent(), received resumed-component: com.seko.quick_usb_example
03-04 17:42:55.719  4957  5106 D GameManagerService: onLooperPrepared(), msg: MSG_APP_RESUME, resumeComponent: ComponentInfo{com.seko.quick_usb_example/com.seko.quick_usb_example.MainActivity}
03-04 17:42:55.719  4957  5106 D GameManagerService: handleResume().
03-04 17:42:55.719  8695  8695 I Utils   : HomePackage : com.sec.android.app.launcher, resumePackageName : com.seko.quick_usb_example
03-04 17:42:55.720  4957  5106 D GameManagerService: handleResume() - secureFolderID: -1301
03-04 17:42:55.720  4957  5106 D GameManagerService: sendFocusOutConditionally() - mPrevNotiPackage: (unknown)
03-04 17:42:55.720  4957  5106 D GameManagerService:   handleResume(). pkgName: com.seko.quick_usb_example, isTunableApp: null
03-04 17:42:55.720  4957  5106 D GameManagerService: notifyFocusInOut(). of pkg: com.seko.quick_usb_example, type: 4, isMinimized: false, isTunableApp: false
03-04 17:42:55.720  4957  5106 D GameManagerService:   notifyFocusInOut(). unexpected mPrevNotiType: -1
03-04 17:42:55.720  4957  5106 D GameManagerService:NetworkLimit: callRelease
03-04 17:42:55.720  4957  5106 D GM:SystemHelper: callReleaseBlock needReleaseBlock:false
03-04 17:42:55.722  5277  5277 V Navbar.Store: getValue(BOOL_SUPPORT_PHONE_LAYOUT_PROVIDER)=false
03-04 17:42:55.722  4957  4995 D SemGameManager: getGMSBinder(), begin
03-04 17:42:55.723  4957  4995 D SemGameManager: getGMSBinder(), successful: true
03-04 17:42:55.723  4957  4995 D SemGameManager: isAvailable(), ret=true
03-04 17:42:55.723  4957  4995 D SemGameManager: isGamePackage(), pkgName=com.seko.quick_usb_example
03-04 17:42:55.723  4957  4995 D SemGameManager: getGMSBinder(), begin
03-04 17:42:55.724  4957  4995 D SemGameManager: getGMSBinder(), successful: true
03-04 17:42:55.724  4957  4995 D GameManagerService: identifyGamePackage. com.seko.quick_usb_example, mCurrentUserId: 0, callerUserId: 0
03-04 17:42:55.724  4957  4995 D PkgDataHelper: getGamePkgData(). com.seko.quick_usb_example
03-04 17:42:55.724  4957  4995 D SemGameManager: isGamePackage(), ret=false
03-04 17:42:55.725  4957  4993 D SehCodecSolutionService: Update top [com.seko.quick_usb_example]
03-04 17:42:55.727  4957  4993 D SSRM:LoadDetectMonitor: PID = 10092, UID = 10188
03-04 17:42:55.727  4957  5198 D PkgPredictorService: com.seko.quick_usb_example go to foreground!
03-04 17:42:55.727  4957  5198 E PkgPredictorService-Collector: record changed bt=0  wifi=1 screen=1
03-04 17:42:55.727  4957  5198 D PkgPredictorService: previous pkgs: com.seko.quick_usb_example,com.seko.quick_usb_example,com.seko.quick_usb_example running pkg: com.seko.quick_usb_example, uid: 0 is system: false
03-04 17:42:55.727  4957  5198 D PkgPredictorService-Collector: save record...
03-04 17:42:55.727  4957  5198 D PkgPredictorService-Collector:  (hour:17 day:6 previous:[com.seko.quick_usb_example, com.seko.quick_usb_example, com.seko.quick_usb_example] activityName:unknown running:com.seko.quick_usb_example userId:0 screenOrientation:1 wifi:1 bt:0 predictTime:1 apkVersion:1.0.0 consumeTime:-1 preloaded:false>)
03-04 17:42:55.727  4957  5198 D PkgPredictorService-NapPreloadController: User using: com.seko.quick_usb_example
03-04 17:42:55.727  4957  5198 I PkgPredictorService-NapClassifier: RF predict version:1646330828849
03-04 17:42:55.728  4957  5198 D PkgPredictorService-NapClassifier: Predict result: 22.0,26.0,25.0, - [0_&_com.seko.apulse_tcm, 0_&_com.seko.infrared_pulse, 0_&_com.seko.usb_experiment]
03-04 17:42:55.737  5277  5277 V Navbar.Store: getValue(BOOL_SUPPORT_PHONE_LAYOUT_PROVIDER)=false
03-04 17:42:55.743  4700  4700 D SurfaceFlinger: Display 0 HWC layers:
03-04 17:42:55.743  4700  4700 D SurfaceFlinger:      DEVICE | 0x7c508157c0 | 0002 | RGBA_1010102 |  400.0    0.0 1600.0 2000.0 |    0    0 1200 2000 | com.android.systemui.ImageWallpaper$_5277#0
03-04 17:42:55.743  4700  4700 D SurfaceFlinger:      DEVICE | 0x7c50816bc0 | 0000 | RGBA_8888    |    0.0    0.0 1200.0 2000.0 |    0    0 1200 2000 | com.sec.android.app.launcher/com.sec[...].activities.LauncherActivity$_5723#0
03-04 17:42:55.743  4700  4700 D SurfaceFlinger:      DEVICE | 0x7cd47b1840 | 0002 | RGBA_8888    |    0.0    0.0 2000.0 1200.0 |   84  140 1116 1860 | Splash Screen com.seko.quick_usb_example#0
03-04 17:42:55.743  4700  4700 D SurfaceFlinger:      DEVICE | 0x7c2fa24500 | 0000 | RGBA_8888    |    0.0    0.0   36.0 2000.0 | 1164    0 1200 2000 | StatusBar$_5277#0
03-04 17:42:55.743  4700  4700 D SurfaceFlinger:      DEVICE | 0x7c50816d00 | 0000 | RGBA_8888    |    0.0    0.0  122.0   67.0 |  841 1933  963 2000 | com.samsung.android.app.cocktailbars[...]arservice.CocktailBarService$_8695#0
03-04 17:42:55.743  4700  4700 D SurfaceFlinger:      DEVICE | 0x7c50814780 | 0000 | RGBA_8888    |    0.0    0.0   72.0 2000.0 |    0    0   72 2000 | NavigationBar0$_5277#0
[... cut ...]
03-04 17:44:11.558  5277  5277 I AODBatteryManager: saveBatteryData : AOD BatteryData [mBatteryLevel=39, mBatteryStatus=DISCHARGING, mBatteryPlugType=NOT_DEFINED, mBatteryPlugged=false, mRemainingChargeTime=0, mBatteryChargingType=0, mBatteryChargerType=NORMAL, mBatteryOnline=NOT_DEFINED, mBatterySwellingMode=NONE, mBatteryProtectMode=false]

03-04 17:44:12.278  4422  4422 D io_stats: !@   8,0 r 974102 31198092 w 474224 18350364 d 213975 66529480 f 741973 818588 iot 2296256 572170 th 102400 0 0 pt 0 inp 0 0 180336.805

03-04 17:44:12.520  5244  5244 I wpa_supplicant: Heartbeat 13419

```

Sunbreak, 2022-03-04:
It seems that you were using a `OUT` endpoint to `bulkTransferIn`

Every endpoint has a direction. Check it before using

https://github.com/woodemi/quick_usb/blob/57fdf0f596ef330de4337837168d7cc20dec6426/lib/src/common.dart#L109-L113

```dart
class UsbEndpoint {
  // Bit 7 indicates direction
  static const int MASK_DIRECTION = 0x80;

  static const int DIRECTION_OUT = 0x00;
  static const int DIRECTION_IN = 0x80;

  final int direction;
}
```

Sunbreak, 2022-03-04:
Sorry. Missunderstanding

It seems that `actualLength` is `-1`

https://github.com/woodemi/quick_usb/blob/57fdf0f596ef330de4337837168d7cc20dec6426/android/src/main/kotlin/com/example/quick_usb/QuickUsbPlugin.kt#L141-L142

According to https://developer.android.com/reference/android/hardware/usb/UsbDeviceConnection#bulkTransfer(android.hardware.usb.UsbEndpoint,%20byte[],%20int,%20int,%20int), `-1` means received nothing

I'll rethrow the logic error from Android API to QuickUsb's Dart API

PR #51 (2022-03-04): Rethrow error from Android API to Dart API
fix #50
closed 2022-03-04 by Sunbreak