woodemi/quick_usb #54: not working on windows

bikram0000, 2022-03-09:
I can get the device list but am not able to open the device, always return false
[image]
[image]
.

Sunbreak, 2022-03-09:
Could you try https://github.com/woodemi/libusb.dart?

How is the open result?

bikram0000, 2022-03-12:
F:\ANDROID\Git Project Plugin\libusb.dart>pub run ffigen
"The top level `pub.bat` command is deprecated. Use `dart pub` instead."
Running in Directory: 'F:\ANDROID\Git Project Plugin\libusb.dart'
Input Headers: [libusb-1.0\libusb.h]
[WARNING]: Skipped Function 'libusb_cpu_to_le16', inline functions are not supported.
[WARNING]: Removed All Struct Members from libusb_bos_dev_capability_descriptor(libusb_bos_dev_capability_descriptor), Flexible array members not s
upported.
[WARNING]: Removed All Struct Members from libusb_bos_descriptor(libusb_bos_descriptor), Flexible array members not supported.
[WARNING]: Removed All Struct Members from libusb_transfer(libusb_transfer), Flexible array members not supported.
[WARNING]: Skipped Function 'libusb_control_transfer_get_data', inline functions are not supported.
[WARNING]: Skipped Function 'libusb_control_transfer_get_setup', inline functions are not supported.
[WARNING]: Skipped Function 'libusb_fill_control_setup', inline functions are not supported.
[WARNING]: Skipped Function 'libusb_fill_control_transfer', inline functions are not supported.
[WARNING]: Skipped Function 'libusb_fill_bulk_transfer', inline functions are not supported.
[WARNING]: Skipped Function 'libusb_fill_bulk_stream_transfer', inline functions are not supported.
[WARNING]: Skipped Function 'libusb_fill_interrupt_transfer', inline functions are not supported.
[WARNING]: Skipped Function 'libusb_fill_iso_transfer', inline functions are not supported.
[WARNING]: Skipped Function 'libusb_set_iso_packet_lengths', inline functions are not supported.
[WARNING]: Skipped Function 'libusb_get_iso_packet_buffer', inline functions are not supported.
[WARNING]: Skipped Function 'libusb_get_iso_packet_buffer_simple', inline functions are not supported.
[WARNING]: Skipped Function 'libusb_get_descriptor', inline functions are not supported.
[WARNING]: Skipped Function 'libusb_get_string_descriptor', inline functions are not supported.
Finished, Bindings generated in F:\ANDROID\Git Project Plugin\libusb.dart\lib\src\libusb.ffigen.dart

Sunbreak, 2022-03-12:
Could you try something like `pub run example/listdevs.dart`, not `ffigen`

You could change the code inside `listdevs.dart` to open the device you list

bikram0000, 2022-03-12:
Building package executable...
Built libusb:listDeviceProducts.
1165:49408 open error: LIBUSB_ERROR_NO_DEVICE
1226:28784 open error: LIBUSB_ERROR_NOT_FOUND
4070:33054 open error: LIBUSB_ERROR_NOT_SUPPORTED
32902:41837 iProduct empty
32903:2730 iProduct empty
deviceProductInfo [MapEntry(1165:49408: ), MapEntry(1226:28784: ), MapEntry(4070:33054: ), MapEntry(32902:41837: ), MapEntry(32903:2730: )]

4070 is a thermal printer and it perfectly working with macOS but not open in windows

Sunbreak, 2022-03-12:
`LIBUSB_ERROR_NOT_SUPPORTED` seems you haven't installed the right driver

bikram0000, 2022-03-12:
right driver for the printer? i think I have installed the correct driver and it's working with test software.

bikram0000, 2022-03-12:
thank you for your help I installed zadig and reinstall the driver, my problem is solved now.

closed 2022-03-12 by Sunbreak

Sunbreak, 2022-03-18:
> I also have the same problem. I ran listdevs.dart. It lists the devices. But still openDevice returns false.

What about `listDeviceProducts.dart` beside `listdevs.dart`?

kalimero-plt, 2022-03-18:
Thank you for your reply. I tried what you said. It gave the error LIBUSB_ERROR_NOT_SUPPORTED. I installed Zadig and reinstall the driver. OpenDevice returned true. Problem solved.