woodemi/quick_usb #55: Communication with USB

kalimero-plt, 2022-03-21:
Hello, I need to send and receive data via USB.
I'm trying to use the bulkTransferIn part, but how do I send the data? Where should I write the data I will send?

Sunbreak, 2022-03-21:
bulkTransferOut

kalimero-plt, 2022-03-21:
flutter: bulkTransferOut error: LIBUSB_ERROR_NOT_FOUND.
bulkTransferOut returns -1

Sunbreak, 2022-03-21:
Could you post `flutter doctor -v`?

kalimero-plt, 2022-03-21:
[√] Flutter (Channel dev, 2.11.0-0.1.pre, on Microsoft Windows [Version 10.0.19044.1586], locale tr-TR)
    • Flutter version 2.11.0-0.1.pre at C:\Users\Enelsis\src\flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision b101bfe32f (5 weeks ago), 2022-02-16 07:36:54 -0800
    • Engine revision e355993572
    • Dart version 2.17.0 (build 2.17.0-69.2.beta)
    • DevTools version 2.10.0-dev.1

[√] Android toolchain - develop for Android devices (Android SDK version 32.1.0-rc1)
    • Android SDK at C:\sdk
    • Platform android-32, build-tools 32.1.0-rc1
    • Java binary at: C:\Program Files\Android\Android Studio\jre\bin\java
    • Java version OpenJDK Runtime Environment (build 11.0.11+9-b60-7590822)
    • All Android licenses accepted.

[√] Chrome - develop for the web
    • Chrome at C:\Program Files\Google\Chrome\Application\chrome.exe

[√] Visual Studio - develop for Windows (Visual Studio Community 2019 16.11.9)
    • Visual Studio at C:\Program Files (x86)\Microsoft Visual Studio\2019\Community
    • Visual Studio Community 2019 version 16.11.32106.194
    • Windows 10 SDK version 10.0.19041.0

[√] Android Studio (version 2021.1)
    • Android Studio at C:\Program Files\Android\Android Studio
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.11+9-b60-7590822)

[√] VS Code (version 1.65.2)
    • VS Code at C:\Users\Enelsis\AppData\Local\Programs\Microsoft VS Code
    • Flutter extension version 3.36.0

[√] Connected device (3 available)
    • Windows (desktop) • windows • windows-x64    • Microsoft Windows [Version 10.0.19044.1586]
    • Chrome (web)      • chrome  • web-javascript • Google Chrome 99.0.4844.74
    • Edge (web)        • edge    • web-javascript • Microsoft Edge 99.0.1150.46

[√] HTTP Host Availability
    • All required HTTP hosts are available

• No issues found!

Sunbreak, 2022-03-21:
If you are working on Windows device, check the following steps

1. `QuickUsb.init` to initialize
2. `QuickUsb.getDeviceList` to get list of devices
3. `QuickUsb.openDevice` to open the right device
4. `QuickUsb.getConfiguration` to get the right config
5. `QuickUsb.claimInterface` to claim the interface
6. `e.direction == UsbEndpoint.DIRECTION_OUT` to check endpoint direction
7. `QuickUsb.bulkTransferOut` to send data

kalimero-plt, 2022-03-22:
It works without error. Thank you for your answers.

closed 2022-03-22 by Sunbreak

Sunbreak, 2022-03-23:
What is a stop bit? Any doc?

You'd better file another issue for new topic

kalimero-plt, 2022-03-24:
A stop bit is a character in asynchronous communication that lets a receiver know that the byte being transmitted has ended.