woodemi/quick_usb #56: Stop bit in communication with USB

kalimero-plt, 2022-03-24:
How can I set the stop bit?

Sunbreak, 2022-03-24:
What is stop bit? Any doc about stop bit?

kalimero-plt, 2022-03-24:
A stop bit is a character in asynchronous communication that lets a receiver know that the byte being transmitted has ended.
I am sending the value 98. It returns 239 for me. That's why I asked for the stop bit.

I reinstalled the driver with zadig but it doesn't load as a port. It installs as a separate USB Serial device. I need to communicate over the port. What can I do?
[image]

Sunbreak, 2022-03-24:
Try https://github.com/woodemi/quick_usb/pull/40 ?

Sunbreak, 2022-03-24:
> A stop bit is a character in asynchronous communication that lets a receiver know that the byte being transmitted has ended. I am sending the value 98. It returns 239 for me. That's why I asked for the stop bit.

I'd like to add it if there is any official API of [libusb](https://github.com/libusb/libusb) available about stop bit

If it is logic upon USB protocol, you'd better implement it in your biz layer