woodemi/quick_usb #57: quick_usb demo can get device list ,but my project get device list empty on macos

ludy520, 2022-04-08:


Sunbreak, 2022-04-08:
Could you try `example/listdevs.dart` inside https://github.com/woodemi/libusb.dart?

```sh
% git clone https://github.com/woodemi/libusb.dart?
% cd libusb.dart
% flutter pub get
% flutter pub run example/listdevs.dart
```

ludy520, 2022-04-09:
example/listdevs.dart 可以获取到，quick_usb的demo也是可以获取到deviceList，然后import到自己的项目，获取到的列表就是空的
[image]

Sunbreak, 2022-04-09:
You need [com.apple.security.device.usb](https://developer.apple.com/documentation/bundleresources/entitlements/com_apple_security_device_usb) in your entitlement file (Debug & Release) according to https://developer.apple.com/documentation/security/app_sandbox

```xml
	<key>com.apple.security.device.usb</key>
	<true/>
```

https://github.com/woodemi/quick_usb/blob/e07a13aa907a73e9a1e934c2671a542f650e1501/example/macos/Runner/DebugProfile.entitlements#L9-L10

https://github.com/woodemi/quick_usb/blob/e07a13aa907a73e9a1e934c2671a542f650e1501/example/macos/Runner/Release.entitlements#L7-L8

closed 2022-04-09 by Sunbreak

Sunbreak, 2022-04-09:
进Flutter Desktop群聊，加微信：wk2311007