woodemi/quick_usb #58: openDevice always returns false for connected usb devices / android phone when running example

jyashi, 2022-04-13:
I might be missing something here.. but running example/lib/main.dart and clicking on openDevices button always returns false for my connected android phone.

I am running the example on windows 10 development.

My phone is in developer mode and usb debugging is on. 

Is there any step to this because i cannot getConfiguration until openDevice returns true.

I am new to usb configuration and using it as a developer so sorry if the question comes out as mundane.

Thank you.

Sunbreak, 2022-04-14:
Could you post `flutter doctor -v` and `flutter devices` when your Android phone connected?

jyashi, 2022-04-14:
Flutter devices 

`Pixel 4a 5G (mobile) • 09221JECB14892 • android-arm64  • Android 12 (API 32)
Windows (desktop)    • windows        • windows-x64    • Microsoft Windows [Version 10.0.19044.1586]
Chrome (web)         • chrome         • web-javascript • Google Chrome 100.0.4896.75
Edge (web)           • edge           • web-javascript • Microsoft Edge 98.0.1108.43`

Flutter doctor 

`[√] Flutter (Channel stable, 2.10.0, on Microsoft Windows [Version 10.0.19044.1586], locale en-GB)
    • Flutter version 2.10.0 at C:\Users\UserName\Documents\flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision 5f105a6ca7 (2 months ago), 2022-02-01 14:15:42 -0800
    • Engine revision 776efd2034
    • Dart version 2.16.0
    • DevTools version 2.9.2

[√] Android toolchain - develop for Android devices (Android SDK version 32.0.0-rc1)
    • Android SDK at C:\Users\UserName\AppData\Local\Android\Sdk
    • Platform android-31, build-tools 32.0.0-rc1
    • ANDROID_HOME = C:\Users\UserName\AppData\Local\Android\Sdk
    • Java binary at: C:\Program Files\Android\Android Studio\jre\bin\java
    • Java version OpenJDK Runtime Environment (build 11.0.10+0-b96-7249189)
    • All Android licenses accepted.

[√] Chrome - develop for the web
    • Chrome at C:\Program Files (x86)\Google\Chrome\Application\chrome.exe

[√] Visual Studio - develop for Windows (Visual Studio Community 2019 16.11.9)
    • Visual Studio at C:\Program Files (x86)\Microsoft Visual Studio\2019\Community
    • Visual Studio Community 2019 version 16.11.32106.194
    • Windows 10 SDK version 10.0.19041.0

[√] Android Studio (version 2020.3)
    • Android Studio at C:\Program Files\Android\Android Studio
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.10+0-b96-7249189)

[√] VS Code (version 1.66.2)
    • VS Code at C:\Users\UserName\AppData\Local\Programs\Microsoft VS Code
    • Flutter extension version 3.38.1

[√] Connected device (4 available)
    • Pixel 4a 5G (mobile) • 09221JECB14892 • android-arm64  • Android 12 (API 32)
    • Windows (desktop)    • windows        • windows-x64    • Microsoft Windows [Version 10.0.19044.1586]
    • Chrome (web)         • chrome         • web-javascript • Google Chrome 100.0.4896.75
    • Edge (web)           • edge           • web-javascript • Microsoft Edge 98.0.1108.43

[√] HTTP Host Availability
    • All required HTTP hosts are available

• No issues found!`

Sunbreak, 2022-04-14:
Formatted with [```]

jyashi, 2022-04-14:
> Formatted with [```]

Thanks for that.

Sunbreak, 2022-04-14:
On my macOS

# flutter-2.2.x devices

```
3 connected devices:

PEUM00 (mobile) • 2d82a62d • android-arm64  • Android 11 (API 30)
macOS (desktop) • macos    • darwin-x64     • macOS 11.2.3 20D91 darwin-x64
Chrome (web)    • chrome   • web-javascript • Google Chrome 100.0.4896.88
```

# flutter-2.2.x run -d macos 

- cd quick_usb/example
- flutter-2.2.x run -d macos
- `init` and `_getDeviceList`

```
venderId: 6353, productId: 11523
```

# flutter-2.10.x pub run example/main.dart of `libusb.dart`

- git clone https://github.com/woodemi/libusb.dart
- flutter-2.10.x pub get
- flutter-2.10.x pub run example/main.dart

```
18d1:2d03 (bus 14, device 16)
 path: 01
```

jyashi, 2022-04-15:
I tried running via github desktop but the project is not configured for windows:

```
PS C:\Users\UserName\Documents\GitHub\libusb.dart> flutter run example/main.dart
No supported devices connected.
The following devices were found, but are not supported by this project:
Windows (desktop) • windows • windows-x64    • Microsoft Windows [Version 10.0.19044.1586]
Chrome (web)      • chrome  • web-javascript • Google Chrome 100.0.4896.75
Edge (web)        • edge    • web-javascript • Microsoft Edge 98.0.1108.43
If you would like your app to run on web or windows, consider running `flutter create .` to generate projects for these platforms.
```

So i tried adding using ```flutter create --platforms=windows libusb``` but it fails to detect the windows files probably because im doing it wrong. Can Windows support for the project also be added from the getgo with tips on how to go further from here?

Sunbreak, 2022-04-15:
`flutter pub run` not `flutter run` on Dart only project

```
flutter pub run example/main.dart
```

Sunbreak, 2022-04-15:
> Can Windows support for the project also be added from the getgo with tips on how to go further from here?

PR welcome!

jyashi, 2022-04-15:
> 

Here is my result 

```
PS C:\Users\UserName\Documents\GitHub\libusb.dart> flutter pub run example/main.dart
06cb:00a2 (bus 1, device 3)
 path: 09
8086:9d2f (bus 1, device 0)
04f2:b604 (bus 1, device 3)
 path: 05
18d1:4ee7 (bus 1, device 6)
 path: 01
8087:0a2b (bus 1, device 1)
 path: 07
```

Sorry i do not understand this output but when i remove my phone usb then path : 01 device disappears so i guess thats the phone. But does that mean that ```openDevice``` should be working as expected?

Sunbreak, 2022-04-15:
If the Android phone is debuggable, `openDevice` should work

You could try the `example/listDeviceProducts.dart`, which calls `openDevice` inside: https://github.com/woodemi/libusb.dart/blob/master/example/listDeviceProducts.dart#L93-L98

And you could debug the Dart code while running through Android Studio/VS Code. Or adding `debugPrint` into `L93-L98`

jyashi, 2022-04-15:
> If the Android phone is debuggable, `openDevice` should work
>
> You could try the `example/listDeviceProducts.dart`, which calls `openDevice` inside: [...]

Okay that helps as a next step. Thanks for that, ill try it out.

AlphaBlend1975, 2022-06-29:
Flutter 3.0.3 Windows 10 Alway return connected=false

C:\Users\Arkom\Desktop\ss\libusb.dart>flutter pub run example/main.dart
 
Failed to build libusb:main:
lib/src/libusb.ffigen.dart:112:15: Error: Type 'pkg_ffi.Char' not found.
  ffi.Pointer<pkg_ffi.Char> libusb_error_name(
              ^^^^^^^^^^^^
lib/src/libusb.ffigen.dart:127:17: Error: Type 'pkg_ffi.Char' not found.
    ffi.Pointer<pkg_ffi.Char> locale,
                ^^^^^^^^^^^^
lib/src/libusb.ffigen.dart:140:15: Error: Type 'pkg_ffi.Char' not found.
  ffi.Pointer<pkg_ffi.Char> libusb_strerror(
              ^^^^^^^^^^^^
lib/src/libusb.ffigen.dart:221:17: Error: Type 'pkg_ffi.Int' not found.
    ffi.Pointer<pkg_ffi.Int> config,
                ^^^^^^^^^^^
lib/src/libusb.ffigen.dart:901:17: Error: Type 'pkg_ffi.UnsignedChar' not found.
    ffi.Pointer<pkg_ffi.UnsignedChar> endpoints,
                ^^^^^^^^^^^^^^^^^^^^
lib/src/libusb.ffigen.dart:925:17: Error: Type 'pkg_ffi.UnsignedChar' not found.
    ffi.Pointer<pkg_ffi.UnsignedChar> endpoints,
                ^^^^^^^^^^^^^^^^^^^^
lib/src/libusb.ffigen.dart:945:15: Error: Type 'pkg_ffi.UnsignedChar' not found.
  ffi.Pointer<pkg_ffi.UnsignedChar> libusb_dev_mem_alloc(
              ^^^^^^^^^^^^^^^^^^^^
lib/src/libusb.ffigen.dart:966:17: Error: Type 'pkg_ffi.UnsignedChar' not found.
    ffi.Pointer<pkg_ffi.UnsignedChar> buffer,
                ^^^^^^^^^^^^^^^^^^^^
lib/src/libusb.ffigen.dart:1152:17: Error: Type 'pkg_ffi.UnsignedChar' not found.
    ffi.Pointer<pkg_ffi.UnsignedChar> data,
                ^^^^^^^^^^^^^^^^^^^^
lib/src/libusb.ffigen.dart:1186:17: Error: Type 'pkg_ffi.UnsignedChar' not found.
    ffi.Pointer<pkg_ffi.UnsignedChar> data,
                ^^^^^^^^^^^^^^^^^^^^
lib/src/libusb.ffigen.dart:1188:17: Error: Type 'pkg_ffi.Int' not found.
    ffi.Pointer<pkg_ffi.Int> actual_length,
                ^^^^^^^^^^^
lib/src/libusb.ffigen.dart:1222:17: Error: Type 'pkg_ffi.UnsignedChar' not found.
    ffi.Pointer<pkg_ffi.UnsignedChar> data,
                ^^^^^^^^^^^^^^^^^^^^
lib/src/libusb.ffigen.dart:1224:17: Error: Type 'pkg_ffi.Int' not found.
    ffi.Pointer<pkg_ffi.Int> actual_length,
                ^^^^^^^^^^^
lib/src/libusb.ffigen.dart:1259:17: Error: Type 'pkg_ffi.UnsignedChar' not found.
    ffi.Pointer<pkg_ffi.UnsignedChar> data,
                ^^^^^^^^^^^^^^^^^^^^
lib/src/libusb.ffigen.dart:1439:17: Error: Type 'pkg_ffi.Int' not found.
    ffi.Pointer<pkg_ffi.Int> completed,
                ^^^^^^^^^^^
lib/src/libusb.ffigen.dart:1475:17: Error: Type 'pkg_ffi.Int' not found.
    ffi.Pointer<pkg_ffi.Int> completed,
                ^^^^^^^^^^^
lib/src/libusb.ffigen.dart:2096:24: Error: Type 'pkg_ffi.UnsignedChar' not found.
  external ffi.Pointer<pkg_ffi.UnsignedChar> extra;
                       ^^^^^^^^^^^^^^^^^^^^
lib/src/libusb.ffigen.dart:2155:24: Error: Type 'pkg_ffi.UnsignedChar' not found.
  external ffi.Pointer<pkg_ffi.UnsignedChar> extra;
                       ^^^^^^^^^^^^^^^^^^^^
lib/src/libusb.ffigen.dart:2223:24: Error: Type 'pkg_ffi.UnsignedChar' not found.
  external ffi.Pointer<pkg_ffi.UnsignedChar> extra;
                       ^^^^^^^^^^^^^^^^^^^^
lib/src/libusb.ffigen.dart:2441:24: Error: Type 'pkg_ffi.Char' not found.
  external ffi.Pointer<pkg_ffi.Char> rc;
                       ^^^^^^^^^^^^
lib/src/libusb.ffigen.dart:2444:24: Error: Type 'pkg_ffi.Char' not found.
  external ffi.Pointer<pkg_ffi.Char> describe;
                       ^^^^^^^^^^^^
lib/src/libusb.ffigen.dart:2721:25: Error: Type 'pkg_ffi.Char' not found.
            ffi.Pointer<pkg_ffi.Char>)>>;
[... 424 lines cut ...]
lib/src/libusb.ffigen.dart:1439:25: Error: 'Int' isn't a type.
    ffi.Pointer<pkg_ffi.Int> completed,
                        ^^^
lib/src/libusb.ffigen.dart:1475:25: Error: 'Int' isn't a type.
    ffi.Pointer<pkg_ffi.Int> completed,
                        ^^^
lib/src/libusb.ffigen.dart:2096:32: Error: 'UnsignedChar' isn't a type.
  external ffi.Pointer<pkg_ffi.UnsignedChar> extra;
                               ^^^^^^^^^^^^
lib/src/libusb.ffigen.dart:2155:32: Error: 'UnsignedChar' isn't a type.
  external ffi.Pointer<pkg_ffi.UnsignedChar> extra;
                               ^^^^^^^^^^^^
lib/src/libusb.ffigen.dart:2223:32: Error: 'UnsignedChar' isn't a type.
  external ffi.Pointer<pkg_ffi.UnsignedChar> extra;
                               ^^^^^^^^^^^^
lib/src/libusb.ffigen.dart:2441:32: Error: 'Char' isn't a type.
  external ffi.Pointer<pkg_ffi.Char> rc;
                               ^^^^
lib/src/libusb.ffigen.dart:2444:32: Error: 'Char' isn't a type.
  external ffi.Pointer<pkg_ffi.Char> describe;
                               ^^^^
lib/src/libusb_base.dart:5:12: Error: Couldn't find constructor 'Long'.
  @pkg_ffi.Long()
           ^^^^
pub finished with exit code

Sunbreak, 2022-06-29:
> Flutter 3.0.3 Windows 10 Alway return connected=false

Please open another issue on https://github.com/woodemi/quick.flutter

closed 2022-10-30 by jyashi