woodemi/quick_usb #60: platformException issue facing only on android 12

vishalideastoimpact, 2022-04-18:
[ERROR:flutter/lib/ui/ui_dart_state.cc(209)] Unhandled Exception: PlatformException(error, com.example.flutter_usb_check: Targeting S+ (version 31 and above) requires that 
one of FLAG_IMMUTABLE or FLAG_MUTABLE be specified when creating a PendingIntent.
E/flutter (11462): Strongly consider using FLAG_IMMUTABLE, only use FLAG_MUTABLE if some functionality depends on the PendingIntent being mutable, e.g. if it needs to be used with inline replies or bubbles., null, java.lang.IllegalArgumentException: com.example.flutter_usb_check: Targeting S+ (version 31 and above) requires that one of FLAG_IMMUTABLE or FLAG_MUTABLE be specified 
when creating a PendingIntent.
E/flutter (11462): Strongly consider using FLAG_IMMUTABLE, only use FLAG_MUTABLE if some functionality depends on the PendingIntent being mutable, e.g. if it needs to be used with inline replies or bubbles.
E/flutter (11462):      at android.app.PendingIntent.checkFlags(PendingIntent.java:375)
E/flutter (11462):      at android.app.PendingIntent.getBroadcastAsUser(PendingIntent.java:645)
E/flutter (11462):      at android.app.PendingIntent.getBroadcast(PendingIntent.java:632)
E/flutter (11462):      at com.example.quick_usb.QuickUsbPluginKt.pendingPermissionIntent(QuickUsbPlugin.kt:18)
E/flutter (11462):      at com.example.quick_usb.QuickUsbPluginKt.access$pendingPermissionIntent(QuickUsbPlugin.kt:1)
E/flutter (11462):      at com.example.quick_usb.QuickUsbPlugin.onMethodCall(QuickUsbPlugin.kt:85)
E/flutter (11462):      at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:262)
E/flutter (11462):      at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:296)
E/flutter (11462):      at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$DartMessenger(DartMessenger.java:320)
E/flutter (11462):      at io.flutter.embedding.engine.dart.-$$Lambda$DartMessenger$TsixYUB5E6FpKhMtCSQVHKE89gQ.run(Unknown Source:12)
E/flutter (11462):      at android.os.Handler.handleCallback(Handler.java:938)
E/flutter (11462):      at android.os.Handler.dispatchMessage(Handler.java:99)
E/flutter (11462):      at android.os.Looper.loopOnce(Looper.java:233)
E/flutter (11462):      at android.os.Looper.loop(Looper.java:334)
E/flutter (11462):      at android.app.ActivityThread.main(ActivityThread.java:8396)
E/flutter (11462):      at java.lang.reflect.Method.invoke(Native Method)
E/flutter (11462):      at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:582)
E/flutter (11462):      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1068)
E/flutter (11462): )
E/flutter (11462): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:607:7)
E/flutter (11462): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:177:18)
E/flutter (11462): <asynchronous suspension>
E/flutter (11462): #2      _MyAppState._has_request.<anonymous closure> (package:flutter_usb_check/main.dart:109:13)
E/flutter (11462): <asynchronous suspension>
E/flutter (11462):

vishalideastoimpact, 2022-04-18:
when i first connect device to app i  got 

lUsbEvent: android.hardware.usb.action.USB_DEVICE_ATTACHED, UsbDevice: /dev/bus/usb/001/007, 1a86-7523 USB Serial, null null
I/flutter (10209):  PlatformException(UsbSerialPortAdapter, Failed to acquire USB device., null, null)

Sunbreak, 2022-04-18:
Could you post `flutter doctor -v` and version of `quick_usb`?

vishalideastoimpact, 2022-04-18:
[√] Flutter (Channel stable, 2.10.3, on Microsoft Windows [Version 10.0.19043.928], locale en-US)
    • Flutter version 2.10.3 at C:\src\flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision 7e9793dee1 (7 weeks ago), 2022-03-02 11:23:12 -0600
    • Engine revision bd539267b4
    • Dart version 2.16.1
    • DevTools version 2.9.2

[√] Android toolchain - develop for Android devices (Android SDK version 32.1.0-rc1)
    • Android SDK at C:\Users\varsha.dukare\AppData\Local\Android\sdk
    • Platform android-32, build-tools 32.1.0-rc1
    • Java binary at: C:\Program Files\Android\Android Studio\jre\bin\java
    • Java version OpenJDK Runtime Environment (build 11.0.11+9-b60-7590822)
    • All Android licenses accepted.

[√] Chrome - develop for the web
    • Chrome at C:\Program Files\Google\Chrome\Application\chrome.exe

[X] Visual Studio - develop for Windows
    X Visual Studio not installed; this is necessary for Windows development.
      Download at https://visualstudio.microsoft.com/downloads/.
      Please install the "Desktop development with C++" workload, including all of its default components

[√] Android Studio (version 2021.1)
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.11+9-b60-7590822)

[√] VS Code (version 1.66.2)
    • VS Code at C:\Users\varsha.dukare\AppData\Local\Programs\Microsoft VS Code
    • Flutter extension version 3.38.1

[√] Connected device (4 available)
    • V2058 (mobile)    • 3081391415000MT • android-arm64  • Android 12 (API 31)
    • Windows (desktop) • windows         • windows-x64    • Microsoft Windows [Version 10.0.19043.928]
    • Chrome (web)      • chrome          • web-javascript • Google Chrome 100.0.4896.88
    • Edge (web)        • edge            • web-javascript • Microsoft Edge 100.0.1185.44

[√] HTTP Host Availability
    • All required HTTP hosts are available

Sunbreak, 2022-04-18:
Formatted with [```]

vishalideastoimpact, 2022-04-18:
I don't get that? ['''']?

Sunbreak, 2022-04-18:
> I don't get that? ['''']?

Please use `Code block`: https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks

Sunbreak, 2022-04-18:
Could you try https://github.com/woodemi/quick_usb/pull/61?

vishalideastoimpact, 2022-04-19:
It's showing pop-up for permission now.
Thanks

closed 2022-04-19 by vishalideastoimpact