woodemi/quick_usb #63: How to keep reading data constantly?

Zopenzop, 2022-05-01:
In my case, data is inputted erratically through the serial port to my android device, thus the bulkTransferIn function doesn't really suite my purpose. Should I run the bulkTransferIn function on a loop or is there a better way to achieve my purpose?

Sunbreak, 2022-05-05:
I'm not sure about the API, could you post an example?

Zopenzop, 2022-05-05:
I was looking for something like an event, I've tried the usb_serial package and it's port object has an event that triggers a function whenever it listens to data.

```
	port.inputStream.listen((Uint8List event) {
		print(event);
		port.close();
	});
```

Sunbreak, 2022-05-06:
That would be a big feature request. I'm not sure I have enough time for cross-platform implementation.

PRs welcome

Zopenzop, 2022-05-06:
Ah sure, just needed to know if there was an existing implementation for the same.

Sunbreak, 2022-05-06:
I'd like to leave it open for future implementation