woodemi/quick_usb #64: Android doesn't require permission to get "manufacturer" or "product"

anik786, 2022-06-07:
Currently on Android, running "getDevicesWithDescription" causes the application to ask for permission for every single device which is a bit tedious. It's also unnecessary, because both the "manufacturer" and "product" can be found without requesting any permission. Only the serial number requires permission.

Therefore can I suggest we remove "UsbDeviceDescription" and just use "UsbDevice" with the fields:

```
class UsbDevice {
   final String identifier;
   final int vendorId;
   final int productId;
   final int configurationCount;
   final String? manufacturer;
   final String? product;
   final String? serialNumber; // requires permission for Android
   
   ...
}
```

And adjust "getDeviceList" in the Android code as:

```
      "getDeviceList" -> {
        val manager = usbManager ?: return result.error("IllegalState", "usbManager null", null)
        val usbDeviceList = manager.deviceList.entries.map {
          mapOf(
                  "identifier" to it.key,
                  "vendorId" to it.value.vendorId,
                  "productId" to it.value.productId,
                  "configurationCount" to it.value.configurationCount
                  "manufacturer" to it.value.manufacturerName,
                  "product" to it.value.productName,
                  "serialNumber" to if (manager.hasPermission(it.value)) it.value.serialNumber else null,
          )
        }
        result.success(usbDeviceList)
      }
```

Sunbreak, 2022-06-07:
PRs welcome!

anik786, 2022-06-07:
Sent a PR, without any breaking changes:
https://github.com/woodemi/quick_usb/pull/65

closed 2022-06-09 by Sunbreak