woodemi/quick_usb #66: Bug: Android not detecting when permission has accepted

anik786, 2022-06-07:
This commit seems to break some code on Android 11 (Not tested on Android 12, but I suspect it breaks some code there). https://github.com/woodemi/quick_usb/pull/61/commits/2cae56e981d54db2bcb605531c6410737f8c0d42

This commit replaced:

```
private fun pendingPermissionIntent(context: Context) = PendingIntent.getBroadcast(context, 0, Intent(ACTION_USB_PERMISSION), 0)
```

to

```
private val pendingIntentFlag = 
  if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.M) {
    PendingIntent.FLAG_IMMUTABLE or PendingIntent.FLAG_UPDATE_CURRENT 
  } else {
    PendingIntent.FLAG_UPDATE_CURRENT
  }

private fun pendingPermissionIntent(context: Context) = PendingIntent.getBroadcast(context, 0, Intent(ACTION_USB_PERMISSION), pendingIntentFlag)
```

However, with the new code I found that once the user accepts the USB permission, the following (Tested on Android 11) always returned the default value:

```
 intent.getBooleanExtra(UsbManager.EXTRA_PERMISSION_GRANTED, false)
```

In other words, the "extras" was not being passed to "onReceive".

However, when reverting "pendingIntentFlag" to 0, it seems to work fine.

However, as of Android 12 (API Level 31), the flags needs to specify as FLAG_IMMUTABLE or FLAG_MUTABLE, which is what this commit tried to fix. 

So my suggestion for now is:

```
private val pendingIntentFlag =
  if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.S) {
    PendingIntent.FLAG_IMMUTABLE or PendingIntent.FLAG_UPDATE_CURRENT // TODO: not sure this works
  } else {
    0
  }
```

I have a feeling this doesn't work on Android 12, but don't have an Android 12 device to test it.

Sunbreak, 2022-06-20:
Fixed by https://github.com/woodemi/quick.flutter/pull/32

yahya-uddin, 2022-06-20:
Yep 👍

closed 2022-06-20 by yahya-uddin